PR2/pr2_robot #234: indigo-devel branch?

wkentaro, 2015-06-18:
Why does this repo have no indigo-devel branch?

wkentaro, 2015-06-18:
where can I send PR which is indigo specific?

TheDash, 2015-06-18:
"Why does this repo have no indigo-devel branch?"

It appears that no code was in development for indigo-devel (hence the name Indigo development). 

Do you want me to create indigo-devel so you can make a PR?

wkentaro, 2015-06-18:
Now I'm trying upgrade the ROS version inside PR2, from hydro to indigo.
I think I should change some codes for this, for example `queue_size` for publishers.
http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers

And yes, I'm willing to make PRs to this repo.

TheDash, 2015-06-18:
Hmm, I think the queue_size stuff only returned warnings for compilers. Was there any problems compiling this by source for you?

wkentaro, 2015-06-18:
I'm just at starting point for the installation, so I'm not sure how many changes are there.
And I needed to change like https://github.com/PR2/pr2_robot/pull/235 for indigo.

wkentaro, 2015-06-18:
> I think the queue_size stuff only returned warnings for compilers.

I'm distracted by the syntax warning about `queue_size` at runtime.
It appears all times..

closed 2016-02-29 by TheDash