PR2/pr2_robot #242: Explicitly set i_clamp_min & i_clamp_max params in default controller_configuration

pgrice, 2016-02-26:
Currently the default arm controller gain config yaml sets only i_clamp.  When the controller loads these PID gains, it uses +/- i_clamp as i_clamp_min and i_clamp_max, and sets i_clamp_min and i_clamp_max on the parameter server.  

If i_clamp is subsequently changed without setting i_clamp_min and i_clamp_max explicitly, the controller loading the PID gains will read the new i_clamp value, but then overwrite min/max with the (not updated) values of i_clamp_min and i_clamp_max still on the parameter server.  The result is a controller running with the incorrect i_clamp_min/max values, which can causes significant changes in behavior and stability...

It seems reasonable to me that the default parameter sets should be updated to set i_clamp_min and i_clamp_max explicitly as an example to others.  This should not alter behavior on start up, but those setting different parameters by copying the yaml file will avoid the above issue.

TheDash, 2016-02-29:
Ahhhh, that's a small bug. Good catch

pgrice, 2016-03-01:
Created Pull Request #244 to set these explicitly.  Note, the PR does not update any version numbers or changelogs.