PR2/pr2_robot #265: ready for release into noetic

davefeilseifer, 2021-03-01:
passes tests during local build in noetic/focal (assuming local install of ros-noetic-eml, staged for next release):

generate_prerelease_script.py   https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml   noetic default ubuntu focal amd64   --custom-repo     pr2_robot__custom-10:git:https://github.com/PR2/pr2_robot.git:kinetic-devel     pr2_ethercat_drivers__custom-11:git:https://github.com/PR2-prime/pr2_ethercat_drivers.git:kinetic-devel     pr2_power_drivers__custom-12:git:https://github.com/PR2/pr2_power_drivers.git:kinetic-devel   --level 0   --output-dir ./

v4hn, 2021-03-01:
Traditionally @k-okada releases all functional PR2 packages.

Kei, please release all the packages @davefeilseifer pointed out in recent issues or make it so they can release them themselves.
I'm grateful someone actually does the job to verify the PR2 for noetic.

hamyyy, 2022-11-21:
curious how I'd go about upgrading to noetic from a fresh install.

I've started this process and so far, running pr2_bringup gives errors when trying to start ethercat communication. pr2_grant seems to not run successfully. I'll provide more info in another issue if this one gains attention. 

There is no information on how I would install everything required for the pr2 to run from scratch. everything seems to be outdated on ROS wiki. Any help and direction on where I should start is highly appreciated.

davefeilseifer, 2023-03-05:
@hamyyy are you trying to run a simulated pr2 or an actual pr2?

hamyyy, 2023-03-06:
@davefeilseifer I am using an actual PR2.

@knorth55 thank you, I got around the problem by running the launch file as root, but that's not ideal for obvious reasons.  I remember messing around with `LD_LIBRARY_PATH` months ago, but to no success, ill try your solution today.
Would you know if it makes a difference to use `pr2-grant` or `ethercat_grant`? as far as I know they both do the same thing except one has pr2 dependencies. I'm also trying to slowly migrate the old PR2 packages to use standard ros packages like `ros_control`. It's definitely possible but it would require a lot of work for very little improvement, but it's something to spend my free time on i guess.

knorth55, 2023-03-07:
> Would you know if it makes a difference to use pr2-grant or ethercat_grant? 

the issues is caused by Linux `setcap` change, and `pr2-grant` and `ethercat_grant` has the same issue.
`ethercat_grant` solved the issue in https://github.com/shadow-robot/ros_ethercat/pull/76 ,
and @v4hn applies the same change to `pr2-grant` in #275 .

I haven't tried to rewrite all `pr2_realtimeloop` with `ros_ethercat`, but I think it is theoretically possible.
It will take a lot of time, but the robot behavior will not change.

v4hn, 2023-03-07:
> I am using an actual PR2

Which lab are you at?

Multiple people started to migrate the PR2 to ros_control at some POINT (including me), but I'm not aware of anything pushed online that works.
It's quite some work for debatable value, but it becomes more important work the ros one/ros2 split.
I'd be happy to try your code on our robot once it works for you.
The other part is gazebo support because once the code is migrated to ros_control quite a bit of deprecated controller code can be thrown away too.

hamyyy, 2023-03-29:
@v4hn Im at University of Technology Sydney (UTS). I'm working on the PR2 as part of my thesis/capstone. I'd be happy to test things as the repos get updated. I hope to fully move to ros_control in noetic by the end of year