PR2/pr2_robot #276: Update calibration script

hamyyy, 2023-03-29:
After [this](https://github.com/PR2/pr2_robot/pull/275) fix, things started to work a little better. Although i was experiencing errors in the following file:

https://github.com/PR2/pr2_robot/blob/80369f9040c795efe5fb7902af4a11e7025822dc/pr2_bringup/scripts/calibrate_pr2.py#L315-L316

The error complains of a missing "Loader" argument to the yaml.load() function. The fix seems to be to add the argument `Loader=yaml.FullLoader`. Otherwise the script raises an error and calibration never occurs.

```python
rospy.set_param(calibration_params_namespace+"/calibrate", yaml.load(open(calibration_yaml), Loader=yaml.FullLoader))
rospy.set_param(calibration_params_namespace+"/hold", yaml.load(open(hold_yaml), Loader=yaml.FullLoader))
```

more info about this error [here](https://github.com/yaml/pyyaml/wiki/PyYAML-yaml.load(input)-Deprecation)

v4hn, 2023-03-29:
@hamyyy please provide pull requests instead of issues for things like this. At least I (as one of the people who can merge patches here) don't use this calibration because we changed the kinematic structure of our PR2.

In your particular solution: I believe you should use `yaml.safe_load` instead of the `FullLoader`.
Afaik the calibration yaml files do not need code execution.

hamyyy, 2023-03-30:
My apologies @v4hn, I'll keep that in mind for next time. I'm relatively new to working on large git projects.
On the topic of calibration, the pr2_calibration packages don't seem to run on noetic (no surprise there). Im slowly getting things working on my PR2. Im curious how you calibrate the cameras to get the pointcloud aligned to the robot's environment

v4hn, 2023-03-30:
At the moment we rely mostly on the robot_calibration framework for hand-eye calibration. In the future we might switch to ATOM.
If you did not modify your PR2, it might be easier to get the old calibration package to run (and send patches). 
Notice that (at least for us) the startup detection of zero-crossings for the arms is notably worse (I've seen 1.5cm EEF distance) in summer due to the joint transmissions and the encoders being *after* the transmissions. This problem was even documented somewhere rather fuzzily in the docs. 
I wrote [this utility](https://github.com/TAMS-Group/tams_pr2_refine_startup_calibration) to get more stable results, and hardware maintenance can definitely help too, but it's a source of error you might want to be aware of when working on calibration that persists between startups.